iavofficial/IAV_Distortion #12: remove Player automaticaly from active Game

iav-DaBi, 2024-05-28:


iav-BjRo, 2024-07-05:
concrete expected behaviour has to be discussed. 
- websocket 'disconnected' event is currently send if the page is unloaded (e.g. tab is closed, but also if the page is reloaded)
 ```
window.onunload = function() {
     socket.emit('disconnected', {player: '{{ player }}'});
 }
```

If the tab is only switched or browser app send to the background, no 'disconnected' event is send. 
Therefore different mechanisms have to be evaluated:

**1. Heartbeat:** Client sends a heartbeat event every x seconds, server checks for heartbeat timeouts.
    - can detect client disconnections from the network / network issues
    - heartbeat (created with javascript on client side) might still be send if the tab is switched or the browser app is switched to the background or device is locked. Behaviour might be different for different OS.
 
**2. Page Visibility API:** Detect if the page is in foreground on client side or not.
    - can detect switching to another tap, app or locking the device.
    - what is the desired behaviour. Should a player waiting in the queue to get a vehicle be forced to stay on the same page/browser app?
        - maybe yes to make sure that he is ready as soon as he gets a vehicle
        - but the should be some sort of timeout since just a automatic lock or screensaver of a mobile device should not remove the player from the queue. Could be very frustrating.

**Proposal for conditions to remove player from queue/vehicle:**
- no heartbeat received for `client_hertbeat_timeout` seconds (make `client_heartbeat_timeout` configurable)
- page (or browser app) is in the background for  `client_pageactive_timeout` (make `client_pageactive_timeout` configurable)
- page is closed (currently no idea how to differentiate between a reload of a page and a page is closed)

iav-BjRo, 2024-07-15:
reviewed with @iav-DaBi

closed 2024-07-15 by iav-BjRo