iavofficial/IAV_Distortion #19: support QR-code for new player

iav-DaBi, 2024-05-29:
As an observer, I want to be invited to drive on a virtual racetrack (via QR code scan), so that I can easily try out the controls with my smartphone and thus become a driver with a low barrier to entry.

iav-BjRo, 2024-06-27:
Example for QR-code generator: 
https://www.qrcode-generator.de/solutions/wifi-qr-code/

iav-DaBi, 2024-07-23:
Access to Distortion Wifi:
https://qrdesigner.app/de?gad_source=5&gclid=EAIaIQobChMImvTJ3r26hwMVFGeRBR3eUA3-EAAYASAAEgKbQfD_BwE&step=1&version=2

Access to Driver UI:
https://qrdesigner.app/de?gad_source=5&gclid=EAIaIQobChMImvTJ3r26hwMVFGeRBR3eUA3-EAAYASAAEgKbQfD_BwE&step=1&version=2

iav-BjRo, 2024-07-26:
ToDo's:
- save screenshots of qr codes in repository
- create new ticket for further clarification for usage of qr code generators (for prettier qr codes)

closed 2024-07-26 by iav-DaBi