iavofficial/IAV_Distortion #24: Connection interruption in a trade fair environment

iav-DaBi, 2024-05-30:
Connection interruption in a trade fair environment with various Bluetooth devices nearby

iav-BjRo, 2024-07-19:
A basic system test has been introduced to test if multiple BLE connections could be established with an Anki car.
This was not possible. After the first connection was established the Anki car could not be found by Bluetooth searches anymore, also connection attempts with the explicit mac address failed with a BleakError that the client could not be found.

In the original Anki Overdrive game, a flashing red light of a car could mean that the car was disabled by being hit from weapons (see [here](https://gaming.stackexchange.com/questions/283776/what-do-the-light-signals-on-the-cars-mean)).
Maybe any of the signals around the system or signal interferences could lead to a situation where a car falsely interprets anything as such a signal which leads to the described behaviour in a crowded place.

iav-DaBi, 2024-07-31:
moved from 2.0.0

iav-BjRo, 2024-08-27:
discussed with @Sandominion and @iav-DaBi :
@iav-DaBi will do some research about enhancing the stability of the bluetooth connection (any external BLE dongle or something else)

iav-BjRo, 2024-09-23:
moved to v3.2.0 since it was not completed. Waiting for feedback from next event in CW39