iavofficial/IAV_Distortion #53: Vehicle speed in km/h

Sandominion, 2024-06-28:
The current speed value (displayed in mm/s) should be converted to km/h, taking into account the scale of the model cars, in order to display a speed of a familiar order of magnitude to the driver.

iav-BjRo, 2024-08-26:
Added config parameter "env_vehicle_scale" to the config file. It can be used to define the scale of the model cars. It is set to a default value of 59. This is the scale for the Anki cars to the "2024 Lamborghini Revuelto": 
`vehicle_scale = length_lambo / lenght_anki = 4947mm / 84mm = 58,89 ≈ 59`

The speed value (v_anki) is measured in mm/s. It is converted to a scaled speed value (v_scaled) in km/h in the driver ui (html) using the following calculation:
`v_scaled = (v_anki / 1000) * 3.6 * vehicle_scale` 

With a maximum speed of the anki cars of `v_anki,max =1200 mm/s` a maximum scaled speed of `v_scaled,max = 255 km/h` is reached.

If a target v_scaled,max shall be reached the needed vehicle_scale can be calculated by:
`vehicle_scale = v_scaled,max,target / ( v_anki,max *3.6 / 1000 )`

iav-BjRo, 2024-08-26:
reviewed with @iav-DaBi and @m-braunschweig

closed 2024-08-26 by iav-BjRo