iavofficial/IAV_Distortion #87: Verbesserte Unterscheidung der Fahrzeuge zwischen realen und rein virtuellen Fahrzeugen

iav-DaBi, 2024-08-02:


iav-BjRo, 2024-08-22:
As discussed with @iav-DaBi for the first attempt, only color variants (not compliant to the IAV style guide) have been created, to have a quick fix to show virtual cars on the track that are easily identifiable and clearly differentiable from the real cars.
An advanced option on how to display the virtual cars on the track has to be discussed with @Sandominion.

- color variants of virtual vehicle images have been created
- high resolution images have been compressed for usage in application
- short instruction on how to compress images have been added in the README in https://github.com/iavofficial/IAV_Distortion/tree/master/doc/Images_high_resolution
- images have been configured in the config file

iav-BjRo, 2024-08-23:
reviewed with @iav-DaBi 

[image]

closed 2024-08-23 by iav-BjRo