iavofficial/IAV_Distortion #223: no uTurn for bots

iav-BjRo, 2025-08-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

iav-BjRo, 2025-08-06:
Might be in src\BotManagement\Bot.py line 39 !?

iav-DaBi, 2025-08-26:
jep. Was in src\BotManagement\Bot.py line 39. issue is done. The bots no longer perform U-turns.